commaai/connect #43: Airbnb lint config proposal

macintoshhelper, 2019-09-10:
Hi, as far as I can tell, there isn't any linting, unless create-react-app has some. Airbnb's React rules are highly recommended and used:

https://www.npmjs.com/package/eslint-config-airbnb

I can open a PR for this if you'd like to use it.

chrisinajar, 2019-09-12:
if you want to add the linter and get things passing in it, i'm totally down

chrisinajar, 2019-10-06:
Resolved by #51

closed 2019-10-06 by chrisinajar

macintoshhelper, 2019-10-11:
:thumbsup: Thanks for resolving the conflicts! I hope the lint config helps with code consistency and readability :) . Is it worth adding some docs on editor integration? https://eslint.org/docs/user-guide/integrations

I've never actually used Create React App, I personally find the manual control of Webpack and Babel really beneficial, especially for SSR or creating custom file loaders.

chrisinajar, 2019-10-11:
You'd definitely be right about CRA 1.x, it's more of a hassle than it's worth, but I find the later versions are very easily customized so instead it's just kind of a template starting point. craco + cra 3.x has been a delight.