commaai/connect #56: Support accounts without a device

andyh2, 2019-11-22:
Explorer currently hangs on loading spinner after logging in with an account that does not have any paired devices. Instead, it should display an appropriate zero-state with pointers on how to buy/pair a device.

maxmetcalfe, 2021-02-20:
It looks like this has been fixed. When you log in without a paired device, you see the "Shared device" in the device list and "Get started with comma two" with a link to the shop.

closed 2021-06-11 by jwooning

simplemachine92, 2022-07-08:
I'm wondering how I can test various features and contribute to the repo without a comma, it doesn't seem this is an option any longer?

jwooning, 2022-07-08:
Besides using the demo there isn't really any option. We would also very much appreciate PRs adding tests for the current state of connect, that should also be possible without a device.