commaai/connect #77: Add (dis)engagement stats to drive(s) page(s)

davetapley, 2020-11-23:
Would you accept a PR to add a time/miles and percentage engaged vs disengaged to the drives and/or individual drive pages?

This information is _sort of_ available via the blue/green distinction on the time line, but it would be nice to have some actual numbers to compare / brag about.

TKassis, 2021-05-25:
I'd like to see this too. General stats for every drive would be nice. Mainly percent time or miles engaged.

TKassis, 2021-05-25:
And on a semi-related note, if the line on the map had a different color between engaged vs not engaged. It'll be nice to see that on the map itself.

vishalkrishnads, 2024-06-19:
I built something like this as my submission for the comma web challenge. [Here](https://vishalkrishnads.github.io/routewatch) it is. It doesn't show the disengagement stats in a timeline, rather, plots it on a map visualizing the routes.