commaai/connect #83: Can't select 'custom' time frame for drives

davetapley, 2021-01-23:
Causes crash on:
- Brave Version 1.18.78 Chromium: 87.0.4280.141 (Official Build) (x86_64)
- Chrome Version 88.0.4324.96 (Official Build) (x86_64)

https://user-images.githubusercontent.com/48232/105564084-da01a200-5cdd-11eb-8394-73bdc4f31f08.mov

jwooning, 2021-05-17:
fixed with a9173eaa7c3e524d6515dacd44640609f8f3ce7a

closed 2021-05-17 by jwooning

davetapley, 2021-05-18:
@jwooning this is still happening for me 😑 

```
Waiting 2227123 for something to change...
2e77efd3816f5347:1 Failed to load resource: the server responded with a status of 404 ()
console.js:35 Stopping worker!
2console.js:35 Error: Can not find utils in context. You either a) forgot to wrap your component tree in MuiPickersUtilsProvider; or b) mixed named and direct file imports.  Recommendation: use named imports from the module index.
    at checkUtils (WithUtils.tsx:14)
    at WithUtils.tsx:26
    at Tg (react-dom.production.min.js:184)
    at bi (react-dom.production.min.js:232)
    at ci (react-dom.production.min.js:233)
    at Di (react-dom.production.min.js:249)
    at Yh (react-dom.production.min.js:248)
    at Ii (react-dom.production.min.js:251)
    at Cd (react-dom.production.min.js:85)
    at HTMLDocument.wrapped (raven.js:376)
_n.<computed> @ console.js:35
overrideMethod @ react_devtools_backend.js:2557
qh @ react-dom.production.min.js:198
Ch.Tn.callback @ react-dom.production.min.js:210
ih @ react-dom.production.min.js:193
hh @ react-dom.production.min.js:193
Vh @ react-dom.production.min.js:217
Zh @ react-dom.production.min.js:220
(anonymous) @ react-dom.production.min.js:250
En.unstable_runWithPriority @ scheduler.production.min.js:18
Fi @ react-dom.production.min.js:250
Di @ react-dom.production.min.js:249
Yh @ react-dom.production.min.js:248
Ii @ react-dom.production.min.js:251
Cd @ react-dom.production.min.js:85
wrapped @ raven.js:376
WithUtils.tsx:14 Uncaught Error: Can not find utils in context. You either a) forgot to wrap your component tree in MuiPickersUtilsProvider; or b) mixed named and direct file imports.  Recommendation: use named imports from the module index.
    at checkUtils (WithUtils.tsx:14)
    at WithUtils.tsx:26
    at Tg (react-dom.production.min.js:184)
    at bi (react-dom.production.min.js:232)
    at ci (react-dom.production.min.js:233)
    at Di (react-dom.production.min.js:249)
    at Yh (react-dom.production.min.js:248)
    at Ii (react-dom.production.min.js:251)
    at Cd (react-dom.production.min.js:85)
    at HTMLDocument.wrapped (raven.js:376)
checkUtils @ WithUtils.tsx:14
(anonymous) @ WithUtils.tsx:26
Tg @ react-dom.production.min.js:184
bi @ react-dom.production.min.js:232
ci @ react-dom.production.min.js:233
Di @ react-dom.production.min.js:249
Yh @ react-dom.production.min.js:248
Ii @ react-dom.production.min.js:251
Cd @ react-dom.production.min.js:85
wrapped @ raven.js:376
6console.js:35 Waiting 2227123 for something to change...
```