commaai/connect #99: Use controlsState instead of carState on the timeline

VirtuallyChris, 2021-05-17:
For cars with a comma pedal in them, ACC is never engaged. This makes the timeline blue for the whole drive even though openpilot was engaged during it.

This issue was half fixed when Greg made the border around the video match controlsState here: https://github.com/commaai/explorer/commit/6211c2f922981164b55f3fbed88fc474cf302960

Now that same fix needs to be applied to the timeline.

[image]

VirtuallyChris, 2021-05-17:
Oops, looks like it's the same issue as https://github.com/commaai/explorer/issues/97. So using controlsState would fix both problems.

trentrand, 2021-06-25:
Looks like the timeline renders based on segment events `type`, which is parsed from the return of `GET /v1/devices/${dongleId}/segments`.

Still trying to uncover how the property `segment.events_json` is populated. A fix would likely involve fixing the PUT so that an LKAS-only driving session properly reports segment events with type "engaged" based on the controlsState.

---

Anyone happen to know how I could pull down and test using the route id mentioned in #97?

adeebshihadeh, 2022-01-31:
I believe this is fixed.

closed 2022-01-31 by adeebshihadeh