commaai/connect #162: Can't view parts of certain drives

jyoung8607, 2021-11-29:
`2cef8a0b898f331a|2021-11-28--14-32-48`, connect can only play about 50% through.

`2cef8a0b898f331a|2021-11-28--17-11-56`, connect can only play about 70% through.

Happens on both PC/Chrome and mobile.

I think the problem here is actually on the back-end, a caching issue of some sort, but there's not a good place for me to file a bug against that. Using the second drive as an example, there are 73 segments uploaded (0-72) but when connect or cabana fetch the `qcamera.m3u8` playlist, only qcameras through segment 47 are returned. The drive otherwise appears to be the proper length, thumbnails are generated, etc. 

These drives were interesting to me, so I had pulled both of them up in connect while the qlog/qcamera upload was still in progress. I think this may have triggered the problem. It feels like the qcamera playlist might have been generated the first time I looked at the drive, while it was only partially uploaded, and now I can't play past that point.

jwooning, 2021-11-30:
Could it be that your browser cached the not yet completed `qcamera.m3u8` file? Both routes do seem to work for me now, I can't seem to find anything weird in the logs when the segments were processed.

jyoung8607, 2021-11-30:
Still a problem for me, with a fresh connect sign-in using a fresh browser (I never use Edge), 47 segment m3u8 for the above 72 segment route:

[qcamera.zip](https://github.com/commaai/connect/files/7628103/qcamera.zip)

[image]

[image]

jyoung8607, 2021-11-30:
Header returned by the backend for the qcamera playlist:

`Last-modified: Mon, 29 Nov 2021 02:48:24 GMT`

Which is right between the uploads of segment 47 and 48:

[image]

I don't have visibility into the backend code, but this really feels like the m3u8 was generated on-demand the first time an API call was made for that drive, and I happened to make that API call before the drive finished uploading, and now it isn't regenerating.

gregjhogan, 2021-12-01:
It must be the CDN edge server closest to you caching the response.  The m3u8 is generated as qlogs are uploaded/processed.  Something has to bust the cache if you load the m3u8 before the route is fully uploaded/processed.

My guess is this segment count query param is supposed to bust the cache (but isn't working):
https://github.com/commaai/connect/blob/912ef4b72d74f3a0f4906cccd5822a2b0651e741/src/components/DriveVideo/index.js#L113

jyoung8607, 2021-12-01:
> It must be the CDN edge server closest to you caching the response. The m3u8 is generated as qlogs are uploaded/processed. Something has to bust the cache if you load the m3u8 before the route is fully uploaded/processed.

Hmm yes agreed! I have a personal VPS based on Dallas and when I fetch the m3ua from there, it runs all the way to segment 72 as it should, but continues to only run to segment 47 when fetched from Cincinnati. Looks like Azure's CDN is capturing my request to the same region whether I come from Spectrum DOCSIS or AT&T 5G.

> My guess is this segment count query param is supposed to bust the cache (but isn't working):

I have a sense this might have been a problem for a long time for both connect and cabana. I've run into things like this with car port development and trying to look at routes before they're fully uploaded. I just never really looked into it before this week; before Joost's excellent work, problems with explorer/connect were the rule rather than the exception.

I don't know what's necessary within Azure to selectively invalidate edge caching of stuff like this. For S&Gs I tacked a `?foo=bar` query string onto it for S&Gs and it didn't seem to change anything. If nothing else, can the relevant files be given a `Max-Age` of an hour or two such that we recover from the issue eventually? It's not like these files are large or high-demand.

jwooning, 2021-12-01:
Indeed the cdn cache setting is set to ignore query strings, @gregjhogan can this be changed this to cache every unique url? Then I'll fix the query param to actually change when the amount of segments processed changes.

pd0wm, 2022-02-16:
@jwooning @gregjhogan Any updates on this?

jwooning, 2022-02-16:
Current behavior is better, but not perfect. The amount of segments is counted from the database, which doesn't necessarily mean that the m3u8 is updated. But for now I think this issue can be closed, at some point we want to generate the m3u8 files on the fly anyway.

closed 2022-02-16 by jwooning

pd0wm, 2022-02-16:
The CDN settings have been updated to use the query params?

gregjhogan, 2022-02-16:
Yes, the CDN settings were fixed.