commaai/connect #176: BlobNotFound

GoldSkis, 2022-01-02:
Attempting to download any camera segment or log from Connect to PC results in this message. (after all files have been requested and uploaded to the servers)

'''
This XML file does not appear to have any style information associated with it. The document tree is shown below.
<Error>
<Code>BlobNotFound</Code>
<Message>The specified blob does not exist. RequestId:6e412e96-401e-0011-2da9-ffa143000000 Time:2022-01-02T07:23:07.5523848Z</Message>
</Error>
'''

jwooning, 2022-01-03:
can you post a route where this happens?

GoldSkis, 2022-01-05:
> can you post a route where this happens?

59755a6574d9ac67|2021-12-31--13-49-02--0

https://cabana.comma.ai/?route=59755a6574d9ac67%7C2021-12-31--13-49-02&exp=1672878023&sig=7pfj4nDKMxDV%2BMoxJFTOtdB8%2FF1UgbZ1aHvyQ8JnpyM%3D&max=17&url=https%3A%2F%2Fchffrprivate.azureedge.net%2Fchffrprivate3%2Fv2%2F59755a6574d9ac67%2F9ddbaade21dd594a5acda80580ff125d_2021-12-31--13-49-02

jwooning, 2022-01-05:
This is a known bug where uploading to a preserved route doesn't work. Unpreserving and then preserving the route again should fix this, if the data is still available.

closed 2022-01-05 by jwooning