commaai/connect #182: Enhancement: Display log / file upload status in connect listing, or in useradmin

JasonJShuler, 2022-01-28:
It is very difficult to find a route with rlogs uploaded. If a column could be added in the useradmin list indicating how many files have been uploaded (ideally something showing the total as well), or a percentage.

This will make it SO much easier to find routes that can be used in Cabana.

gregjhogan, 2022-01-28:
raw files are only kept for a max of 14 days, unless you preserve a route, and preserved routes are always at the top of the list in useradmin

note that when you preserve a route, it only preserves the files that exist at the time you click the preserve button

JasonJShuler, 2022-01-29:
> raw files are only kept for a max of 14 days, unless you preserve a route, and preserved routes are always at the top of the list in useradmin
> 
> note that when you preserve a route, it only preserves the files that exist at the time you click the preserve button

I have many devices shared with me, and presently we cannot initiate log uploads on these. There are often many very similar looking drives, and I am having a needle in the haystack sort of problem finding the one or two that actually have enough rlogs to run cabana. So even within a 14-day period this would be very helpful. Even just an indicator that Cabana is available in the list would be fantastic!

I haven't looked at the code yet to see how difficult this would be to implement - if I get some time (and you think it could be valuable) I'm open to trying

gregjhogan, 2022-01-30:
Yeah, just trying to help you out under the current constraints. If you have each person preserve a fully uploaded route you will always have one at the top that you can immediately open in cabana.