commaai/connect #203: readonly users should be able to request uploads

jyoung8607, 2022-03-22:
C2/C3 owners who have shared their devices readonly with another user are doing it because they need remote support. Other than very simple things like UDS fingerprint lookups, these remote support needs usually require full logs to be uploaded.

Now that full logs can only be uploaded on-demand, each drive we want to look at needs to be manually uploaded, and today that can only be done by the actual device owner. This means we need new, novice users to constantly take extra actions.

From a privacy perspective, there's no difference between the full logs/full videos and the decimated logs/low quality videos, and readonly users can see the full versions once uploaded. The only thing we can't do is trigger the upload action.

JasonJShuler, 2022-05-18:
Agreed - if it could be added to connect, or if uploads could be initiated via SSH from the device to make it more obscure.

I wonder if a solution would be to implement a readwrite role - right now it seems there is only owner and readonly. The the readwrite users could have permission to upload logs, etc.

jyoung8607, 2024-06-18:
@adeebshihadeh per our discussion, it would be fine to implement this for rlogs only, no cameras. Will add a poll later.

jyoung8607, 2024-06-20:
[image]