commaai/connect #252: Request for improvement of GPS coordinates inconsistency (coordinates of "COMMA CONNECT WEBPAGE and GOOGLE MAP")

haram-KONA, 2022-09-12:
When using navigation in many countries including south Korea, it is almost impossible to search by "address or street name". 
In order to improve this condition, if we search for a destination in "GOOGLE MAP" and use the "GPS coordinates of the point", 
users in various countries are likely to use the improved navigation. 

problem is that when we search for a destination point using GPS coordinates in "COMMA CONNECT WEBPAGE", 
there is a difference from the "GPS coordinates" checked in "GOOGLE MAP". 

If comma.ai can match this part exactly, we can use navigation well in countries where cannot search for "addresses or street names".

It would be good if you check it once and take a look.

https://discord.com/channels/469524606043160576/905963231879757894/1018543435688456283

incognitojam, 2022-09-16:
Can you give an example of some coordinates so that it can be reproduced?

haram-KONA, 2022-09-16:
> 
To select a destination, use a phone app or other map to find the GPS coordinates of the destination i want to go to.
[image]

haram-KONA, 2022-09-16:
The place I found is about 200KM away from where I live, and the GPS coordinates are identified as "37.866238 128.848528".
Again, these coordinates are substituted into the comma connect webpage.
[image]

haram-KONA, 2022-09-16:
Then it shows the wrong location (37.558860 126.999890) like below. If i open google map using "open in map" here, the wrong place and coordinates are shown as shown below.
[image]

haram-KONA, 2022-09-16:
In this state (google map), if I input the original coordinates again (37.866238 128.848528), it exactly matches the destination I first found.
[image]

[image]

haram-KONA, 2022-09-16:
In the end, if i search using GPS coordinates on the comma connect webpage, i can search, but it actually finds and displays the wrong place.