commaai/connect #277: Replace raw files popup with a table

adeebshihadeh, 2022-10-18:
In the current UI, clips and raw segment file download are treated pretty equally, which is pretty confusing since most user just want a clip. At the same time, I think we can improve the raw file download experience. I propose making the file popup into a useradmin-style table with all the segments.

[image]

criticic, 2024-05-30:
I'd like to work on this

RossGGG, 2024-06-01:
I started tinkering with this one a few days ago.  I was wondering if you could offer some clarification, though.  Did you mean replacing the Log Data download button in the Files popup menu with a modal window that provides the table?

[image]

closed 2024-06-25 by adeebshihadeh