commaai/connect #330: Button to allow quick filling out a GitHub bug report to openpilot from connect

nelsonjchen, 2022-12-04:
https://github.com/isaacs/github/issues/99#issuecomment-443976302

Maybe gate it behind some checks:

* running openpilot
* running comma codebase
* not dirty

nelsonjchen, 2022-12-04:
Wonder if this could work with a template: https://github.com/commaai/openpilot/issues/new?assignees=&labels=car%2Cbug&template=car_bug_report.yml

incognitojam, 2023-01-03:
I think driving experience bug reports are better shared in the discord #openpilot-experience channel

nelsonjchen, 2023-01-12:
Looks like some enhancements have been done to that. I'll close this for now.

On another note, this was released in the meantime: https://github.blog/2023-01-09-github-discussions-just-got-better-with-category-forms/  You can have forms for discussions now apparently.

closed 2023-01-12 by nelsonjchen