commaai/connect #345: navigation component needs refactoring

hiddegrei, 2023-02-04:
We will investigate whether we can apply refactoring in an existing code base. Let's take a look at the OpenPilot open source project. Openpilot is developed by Comma AI and focuses on developing software for self-driving cars. We will see if we can apply refactoring in the code base of the front-end of the Commai AI connect app. The github page of the connect app is https://github.com/commaai/connect. When we look at the various components in the app, something quickly stands out. We see that there are so-called 'bloaters' and 'dispensables'. For example, in the component responsible for navigation, we can see that at the beginning of the file there are 240 code lines for the css styles of the component's front-end. After these css styles follows a list of 40 parameters. This makes the code very cluttered and it is not possible to focus directly on the actual code. If we look at the methods in the component, we see that there are 40 different methods, some of which have the same function, such as rendering a certain front-end or updating the device used, for example. We can put the methods responsible for rendering in a separate class to make the code clearer. We can also put the css styles in a separate css file to improve clarity.

hiddegrei, 2023-02-04:
Great job on analyzing the code base of the OpenPilot project and highlighting the potential issue with the 'bloaters' and 'dispensables' in the front-end of the Comma AI Connect app. Your proposal of refactoring the code by separating the CSS styles and methods in the navigation component is well thought out and can definitely improve the readability and maintainability of the code.

Refactoring is a crucial part of software development and is often overlooked. By improving the internal structure of the code, it not only makes it easier for developers to work with, but also helps to prevent potential issues and bugs in the future. Your post serves as a great reminder of the importance of code refactoring and the positive impact it can have on a code base.

It's also great to see that you have taken the initiative to apply refactoring in an existing code base, and to share your insights with others. I'm sure this post will be of great help to developers who are new to the OpenPilot project, and will also encourage more contributions to the project.

hiddeG, 2023-02-04:
Great job on analyzing the code base of the OpenPilot project and highlighting the potential issue with the 'bloaters' and 'dispensables' in the front-end of the Comma AI Connect app. Your proposal of refactoring the code by separating the CSS styles and methods in the navigation component is well thought out and can definitely improve the readability and maintainability of the code.

Refactoring is a crucial part of software development and is often overlooked. By improving the internal structure of the code, it not only makes it easier for developers to work with, but also helps to prevent potential issues and bugs in the future. Your post serves as a great reminder of the importance of code refactoring and the positive impact it can have on a code base.

It's also great to see that you have taken the initiative to apply refactoring in an existing code base, and to share your insights with others. I'm sure this post will be of great help to developers who are new to the OpenPilot project, and will also encourage more contributions to the project.

nickR3, 2023-02-04:
Your blog post highlights some of the challenges in the front-end code of the Comma AI Connect app in the OpenPilot project. Refactoring is an important aspect of software development and can help improve the code structure and make it easier to work with.

incognitojam, 2023-02-07:
Clear and readable merge requests to address specific issues are more than welcome, but vague issues asking for refactors are not useful.

closed 2023-02-07 by incognitojam