commaai/connect #367: connect: missing segment alert doesn't reappear

sshane, 2023-06-23:
https://connect.comma.ai/d5f0cbd9ceba7d0c/1687485479920/1687485661872

When clicking on a segment with no qcameras, `This video segment has not uploaded yet or has been deleted.` correctly shows. Then you click off that segment and it plays again, fine. Then when you click back to the segment with missing files, it just shows that it's loading indefinitely. It should show that same error again.

adeebshihadeh, 2023-07-06:
Another case: https://connect.comma.ai/aa367395e351d0b2/1688655126537/1688657014434

sshane, 2023-07-06:
those qcameras are missing, except the first

incognitojam, 2023-09-11:
No, I think it involves changes to Hls.js/react player since it isn't reporting the fetching the same mpeg-ts segment again. I'll be looking at this after the prime stuff