commaai/connect #394: Uploaded clips not able to be seen

Zack1010OP, 2023-07-09:
After uploading my clips on Comma Connect, when i try to view my clips it just keeps showing the loading icon. It's been 2 days already. This is so frustrating, i've ALWAYS had problems with Comma's uploaded clips, only once did it work after @sshane helped escalate and fix it previously

i don't know what other information you need, but do let me know and i'm happy to provision all relevant details.

[image]

incognitojam, 2023-07-09:
- Can you share the dongle ID for the clips you are trying to view?
- What browser and operating system are you using?
- If you know how, open the browser developer console and take a screenshot

Zack1010OP, 2023-07-09:
Dongle ID: ecb854b94fdfa4d5
Browser: Chrome (Mac) & iPhone App (iOS)
here is the screenshots showing the errors
[image]

Zack1010OP, 2023-07-09:
if it would help, could you kindly give me the instructions how to open up the "non-minified dev environment" do i can give you the "full errors and additional helpful warnings". thanks!

incognitojam, 2023-07-09:
Can you click on the link `https://reactjs.org/docs/error-decoder.html...` and share the full URL with me?

incognitojam, 2023-07-09:
Actually, fixed in 085506f9db315e5af2b147d8a97b81ee262daaf5. The fix will be live shortly. Thanks for the report!

closed 2023-07-09 by incognitojam

Zack1010OP, 2023-07-09:
[image]

Zack1010OP, 2023-07-09:
ok great, thanks incognitojam, much appreciated!!