commaai/connect #402: video playback broken in Firefox

incognitojam, 2023-07-13:
Video buffering and stuttering continuously for 30 seconds.

Recovers immediately after pausing and resuming.

briznad, 2023-08-01:
I can't reproduce using Firefox 116.0 (64-bit) on a 13-inch, M1, 2020 MacBook Pro running MacOS 13.3.1 (a) (22E772610a) with this randomly chosen route:
https://connect.comma.ai/45ab8705b36f4c31/1690849007550/1690850399652

Can you share Firefox, device, and route/video details to help narrow down the issue?

incognitojam, 2023-08-01:
@sshane how did you reproduce this?

sshane, 2023-08-01:
you can't reproduce this? it's super easy

https://github.com/commaai/connect/assets/25857203/c115b70c-5c4b-4836-a707-d68430c398ea

incognitojam, 2023-08-14:
I can't reproduce it by doing that! Just spent a few minutes trying...
[image]

incognitojam, 2023-08-14:
I'm hoping https://github.com/commaai/connect/pull/425 improves this

BBBmau, 2024-05-15:
Stutter still happens when using firefox, could be something minor since it works fine on chrome