commaai/connect #424: timeline segments skip on certain route

sshane, 2023-08-07:
https://connect.comma.ai/db55e16f25039cb3/1625614679398/1625615630552

https://github.com/commaai/connect/assets/25857203/b44ae94f-2465-44e0-b3f3-0315d77ff0c3

seems to still play the video fine though

incognitojam, 2023-08-07:
Not clear if it's related but there are lots of missing `events.json` files. Not sure why those wouldn't be missing, but I think this route is too old to be able to check logs.. 
[image]

sshane, 2023-08-07:
oh so the timeline missing on old routes is not expected? I notice that a lot on old routes

incognitojam, 2023-08-07:
I don't know why only random segments would be missing. We might have a retention policy for these files...

closed 2023-09-11 by incognitojam