commaai/connect #427: Replace carousel lib with CSS

incognitojam, 2023-08-22:
`react-responsive-carousel` is only being used for the [device snapshot images](https://github.com/commaai/connect/blob/master/src/components/DeviceInfo/index.jsx#L376) on mobile. We can implement snapping easily with CSS: https://developer.mozilla.org/en-US/docs/Web/CSS/scroll-snap-type

aurnik, 2023-12-23:
Is there a recommended way to test this without a device and screenshots? I wrote a quick fix but still need to test: https://github.com/aurnik/connect/commit/1062d5d85518bff1123c697bda9a88c0fa889d04

adeebshihadeh, 2024-05-26:
@aurnik I'm going to setup a real demo account for this reason. interested in working on #474 if I can send you over a token for that demo account?

PR #475 (2024-05-26): Components/DeviceInfo: Replaced `react-responsive-carousel` w/CSS
**TL;DR**  This replaces carousel lib with CSS which fixes #427 

## Scroll bar always shows (except for Firefox & Safari)
By default `scroll-snap-type` hides scroll bar if user isn't using it, I decided to set it to be always on because pagination circles are not there anymore to indicate that there is another photo to be seen. I also made scroll bar a bit more noticeable by making it taller so users can find out quicker that it is scrollable/swipe-able. I noticed Firefox and Safari does not seem to support the scrollbar to alw [... cut]
closed 2024-05-26 by adeebshihadeh