commaai/connect #429: I cannot register my home or office with C3X, it doesn't register.

haram-KONA, 2023-09-02:
As shown in the attached video and picture, when I try to register my home or office, "SAVE AS" changes to "SAVED" in "comma connect" and it needs to be registered, but it does not work. I don't know if I'm doing it wrong, but please take a look. I don't know if this is a problem only in Korea.

https://github.com/commaai/connect/assets/88036668/7020925d-263e-49ae-972b-6879dbed4dd1

[image]

haram-KONA, 2023-09-02:
my comma device is C3X , live in South Korea, DongleID "1618132d68afc876"

jyoung8607, 2023-09-02:
Also of interest, the "navigate" button is missing, so the address may not be understood or it may think it's unreachable.

haram-KONA, 2023-09-02:
[image]

haram-KONA, 2023-09-06:
Ah~~~~~ 

Is this problem only happening to me?

Could solve this quickly? 
Unable to set navigation destination!!! 
Please check it quickly.

incognitojam, 2023-09-06:
Do you know if this worked in the past? I see you also have access to another device in connect. Does it also happen when trying to set the destination for another device?

haram-KONA, 2023-09-06:
> Do you know if this worked in the past? I see you have access to another device in connect. Does it also happen when trying to set the destination for another device? No worries if you cannot test this if you don't have prime on another device.

You are saying that i used C3 before using C3X. 
The device C3 is currently unpaired. 
This phenomenon did not occur when using C3 in the past. 
It seems to only occur in C3X, but I don't know if it's actually a problem in C3X or the comma cloud.
Or, I don't know if it's a problem in Korea, outside of the US or Europe.
It's just that things that were set up and saved well in the past don't work at all now. 
Currently, I am registered as LitePrime.(C3X)

haram-KONA, 2023-09-06:
this is C3

haram-KONA, 2023-09-06:
One thing to keep in mind is that before using C3X, I used two C3 devices, and both devices were subscribed to LitePrime, and both devices were registered well and the destination settings were fine. And after purchasing the C3X, I canceled the prime on both C3s and they were unpaired.

haram-KONA, 2023-09-06:
I tried it in Windows Explorer, my Android phone, and FireFox on Linux mounted on VMware, but they all failed the same way.     blabla ~~~

incognitojam, 2023-09-06:
Thanks, it's helpful to know that it worked in the past, regardless of whether it was a comma 3/3X!

sshane, 2023-09-11:
@incognitojam any update? Any logs in sentry?

haram-KONA, 2023-09-15:
Does this take a long time to resolve?

haram-KONA, 2023-09-26:
Why is this state not being solved? 
For some reason? 
Or is it a problem that cannot be solved? 
Please let me know.

There doesn't seem to be any action, there doesn't seem to be any specific information, and I'm extremely curious.

haram-KONA, 2023-09-26:
The strange thing is that if i specify the "destination" as GPS coordinates on the "Comma_Contact web page" and save it, it works very well.
In Korea, it is 100% impossible to find a destination using a place name or address, so i have to use GPS coordinates. 
However, this method(GPS Cord) cannot be used at all because the actual coordinates and the point indicated on the "comma connect web page" are completely different. 
I also raised this as a "comma connect issue" about a year ago, but I haven't heard any action or related information since then.