commaai/connect #440: Viewing video on mobile is broken

nelsonjchen, 2023-10-29:
Can recreate by popping open inspector, setting it to emulate an iPhone SE viewport, and pasting a URL like:

https://connect.comma.ai/fe18f736cb0d7813/1698515982559/1698516042552

in. 

[image]

Seems others see this too: https://www.reddit.com/r/Comma_ai/comments/17j54ay/comma_connect_unable_to_view_videos/

nelsonjchen, 2023-10-30:
Seems to happen on my M1 Macbook. Wonder if this generally the case for Apple devices.

adeebshihadeh, 2023-10-30:
Thanks for the good repro steps! Fix should be pushed soon.

adeebshihadeh, 2023-10-30:
This doesn't appear to be a real issue, at least the one with the SE in Chrome. React-player is identifying it as iOS, which doesn't support hls.js and instead uses its own HLS player, which Chrome doesn't have. So this particular case is an issue with Chrome faking Safari.

adeebshihadeh, 2023-10-31:
Got a reliable repro for the M1 issue?

nelsonjchen, 2023-10-31:
Not from me. I think you ID'd a deficiency in my test and how it's not realistic. Replied to the user to come over here who responded "I’m using Windows 11 on desktop and iOS 17 on mobile." : https://www.reddit.com/r/Comma_ai/comments/17j54ay/comment/k7403za/?utm_source=share&utm_medium=web2x&context=3

adeebshihadeh, 2023-10-31:
I'm following the Reddit thread, so I'll close this.

closed 2023-10-31 by adeebshihadeh

devimauz, 2023-11-04:
@adeebshihadeh It's an OP bug not Connect bug. It's not recording the route this is why it's wont upload it. 4ffdd3a6a974b54b|2023-11-04--18-17-31--0