commaai/connect #467: Search returns no results

BBBmau, 2024-05-16:
Before when typing it would provide you with suggestions. This doesn't happen anymore and just gives no results found.

If this is a server issue then it would be worth showing some kind of message so that users know the reason. 

[image]

BBBmau, 2024-05-16:
Came across [this](https://stackoverflow.com/questions/66150974/here-api-autosuggest-occasionally-returns-401-unauthorized-when-proxied-through) that relates to this bug

The response is a 401 unauthorized error. im assuming that connect is on a freemium plan with them and they just hit their rate limit?

ebellumat, 2024-05-16:
More details here: https://github.com/commaai/connect/issues/468

BBBmau, 2024-05-16:
https://github.com/commaai/connect/commit/0216a272064c202e15a13283358d1f28543f60d6 solved the issue