commaai/connect #473: [$500 bounty] Adjust drive definition from unix time range to an openpilot route

adeebshihadeh, 2024-05-25:
The goal is to change the definition of a viewable "drive" in connect from a unix timestamp range to simply be an openpilot route/log ID, `fa7cd420146956ea/1712332179093/1712332850533` -> `fa7cd420146956ea/00000028--7e47f53956`. This was originally done so that multiple drives could be showed together, but this turns out to introduce a bunch of complexity and also not be useful. 

Old: `connect.comma.ai/<dongle_id>/<unix_start_time>/<unix_end_time>`
New: `connect.comma.ai/<dongle_id>/<openpilot_log_id>`

Guidelines
* use relative time where possible in components
* need to do something about old connect links, e.g. go to the drive list and set the time range that's in the URL

There's a (significant) start at it here: https://github.com/commaai/connect/pull/458

ManyRios, 2024-05-29:
Hey, Can I help with this one?

adeebshihadeh, 2024-05-31:
Locked to @0x7B5. Let me know if you have any questions.

0x7B5, 2024-05-31:
@adeebshihadeh is there another demo account I can use with more routes on it?

adeebshihadeh, 2024-05-31:
> @adeebshihadeh is there another demo account I can use with more routes on it?

Yes, we set one up for https://github.com/commaai/connect/issues/474

0x7B5, 2024-06-02:
With zooms, do you want to keep urls params or remove in that situation as well? 

ie.: <dongle_id>/<log_id>/<relative_start_time>/<relative_end_time>

adeebshihadeh, 2024-06-02:
> @adeebshihadeh With zooms, do you want to keep urls params or remove in that situation as well?
> 
> ie.: <dongle_id>/<log_id>/<relative_start_time>/<relative_end_time>

No zoom should just be log id, and zoom should be relative start and end seconds.

sshane, 2024-06-03:
Keep in mind how this works if you zoom to share a section of a route as it's uploading, you don't want that selected section to change as the end time changes.

0x7B5, 2024-06-03:
@sshane The first part of that should be addressed with relative offsets, the second part of expanding as it uploads I'll have to look into.

0x7B5, 2024-06-04:
@adeebshihadeh I sent you a message on discord

PR #486 (2024-06-18): Adjust drive definition from unix time range to openpilot route
fixes #473
closed 2024-06-18 by adeebshihadeh