commaai/connect #474: Use a real account for the demo experience

adeebshihadeh, 2024-05-25:
The demo experience is one route with some hardcoded API responses and "if demo" in a bunch of places.

Instead, the demo should be a real user and hit all the same code paths a normal user does.  

It'll be a better demo experience, and it should enable external contributors without a device to more effectively contribute.

ugtthis, 2024-05-27:
Would also be nice for the demo account to have

- Ability to do snapshots, current demo experience "take snapshot" button is disabled 
- An option or just by default, make the demo account have about 5-10 trips so that we can see how certain designs affect the trip list. Currently demo account only has 1 trip by default

sshane, 2024-05-30:
Here is a token to access a read only demo device in connect! Just place it under `authorization` in your local storage to try it out. You may need to enter an incognito session. This issue is now for moving our demo experience over to using this account/token. You should remove a bunch of `if isDemo` lines. @adeebshihadeh may want to make this a bounty

`eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJleHAiOjE3NDg1ODI0NjUsIm5iZiI6MTcxNzA0NjQ2NSwiaWF0IjoxNzE3MDQ2NDY1LCJpZGVudGl0eSI6IjBkZWNkZGNmZGYyNDFhNjAifQ.g3khyJgOkNvZny6Vh579cuQj1HLLGSDeauZbfZri9jw`

PR #485 (2024-06-03): Use a real account for the demo experience
Fixes #474
Puppeteer tests are yet to be changed
closed 2024-06-03 by adeebshihadeh