commaai/connect #520: connect links outside date range do not load

sshane, 2024-06-19:
Just shows the drive list

vishalkrishnads, 2024-06-20:
How exactly can this be reproduced? I tried selecting a date range, and then pasting a new drive link in the address bar. It seems to be loading fine.

adeebshihadeh, 2024-06-20:
> How exactly can this be reproduced? I tried selecting a date range, and then pasting a new drive link in the address bar. It seems to be loading fine. The drive link was outside the date range.

To repro with the demo routes, you'll probably want to force the range to be like 1 day.

vishalkrishnads, 2024-06-20:
I selected the start & end dates as May 22, 2024 and pasted [this drive link](https://connect.comma.ai/1d3dc3e03047b0c7/000000dd--455f14369d), which is on May 23, 2024.  I also tried with [this one](https://connect.comma.ai/1d3dc3e03047b0c7/000000d1--9e0f877e15) from May 20, 2024. Still not seeing the issue, everything is loading fine.

sshane, 2024-07-02:
API with new input route_str to endpoint routes_segments has been deployed to staging URL: https://api.aks.comma.ai/

closed 2024-07-12 by sshane