commaai/connect #532: load more fails sometimes when opening a device from a route

sshane, 2024-06-24:
https://github.com/commaai/connect/assets/25857203/eaa05d19-29f4-40a7-8291-8b80e53b8406

yaodingyd, 2024-06-24:
initial tests didn't really reproduce the issue. I think it might be another weird race condition. I'll spend some time first to get a consistent way to reproduce the bug

yaodingyd, 2024-07-01:
I still have trouble to reproduce, actually not able to get one yet. do you have any pointers?

sshane, 2024-07-01:
Happens even on a new browser without extensions on the demo device.

- Open a new tab and paste a link to a recent route
- Click the device on the left
- Doesn't load 2nd page
- Refresh, it does

https://github.com/commaai/connect/assets/25857203/d854bb7e-599e-4c7e-9d6d-9df5cb3b067d