commaai/connect #536: Remove segment_start_times usage (and friends)

adeebshihadeh, 2024-06-26:
Need to remove `segment_start_times`, `segment_end_times`, and `segment_numbers` since we'll no longer have this information soon.

They're primarily used for file uploads. We'll replace them with:
* make the following assumptions
  * route starts at segment 0
  * segments are 60s
  * segments aren't missing in the middle of the route
* add an "Upload all for route" button that requests uploads for `range(0, max_seg_number+1)`. this is a fallback for cases where the assumptions don't hold true

With those assumptions, the file upload should be mostly unaffected.

neehhaa06, 2024-08-08:
Hello @adeebshihadeh : )

fix: #536

> I think we should implement this process to cleanup.

> 1. Remove segment time variables and calculate segment indices using a 60-second duration. 
> 2. Implement segment upload functions and a button for bulk uploads, then test the new logic.

```
function getSegmentIndex(time) {
    return Math.floor(time / 60);
}
```

### Implement Upload:

```
function uploadSegment(index) {
    // Upload logic
}
```

### Upload All

```
function uploadAllForRoute(maxSegmentNumber) {
    for (let i = 0; i <= maxSegmentNumber; i++) {
        uploadSegment(i);
    }
}
```

### Add Button:

```
const uploadButton = document.createElement('button');
uploadButton.textContent = 'Upload all for route';
uploadButton.onclick = () => uploadAllForRoute(maxSegmentNumber); // Define maxSegmentNumber
document.body.appendChild(uploadButton);
```

polooner, 2024-11-11:
looks like its completed by https://github.com/commaai/new-connect/pull/77

closed 2024-11-11 by adeebshihadeh