commaai/connect #548: Routes have incorrect starting point and reporting 2x actual mileage.

Memberoffoxhound, 2024-08-03:
- Comma 3 0.9.7-release Dongle 49734429dda9bbe7
- 2x mileage issue began approximately 7/14/24
- Secondary issue of incorrect starting location began approximately 7/16/24
- Both issues continue today on every route
- Troubleshooting has included rebooting device and reinstalling openpilot. I have not reflashed or tried to unpair/repair in connect. 

Longer description:
On routes from 7/14/24 connect seems to have around double the miles reported for each route. Even stranger starting on 7/16/24, all routed have a starting location in Chesterfield, regardless of my actual location. Chesterfield is where I work and 40 miles from home. 

I am on stock  release and did attempt to reinstall openpilot. I haven’t done anything too extreme yet such as reflashing my comma 3 and I haven’t seen any mention of this anywhere else. The route below should be approximately 12 miles but shows 24. I started in Godfrey, IL despite it showing a starting point 40+ miles away in Chesterfield. 

https://connect.comma.ai/49734429dda9bbe7/00000000--3324219164

[image]

Memberoffoxhound, 2024-08-13:
Small update. Decided to re-flash the device tonight using the web utility and switch from stock to a fork. Both issues persist using sunnypilot. Moving back to stock release at this time. 

TL;DR
- Re-flashed comma3 from web utility
- Switched from stock to sunnypilot fork
- Both starting point and 2x mileage issues remain
- Switching back to stock

Memberoffoxhound, 2024-08-13:
Re-paired device on connect. No change.

m01001100, 2024-09-07:
I have the double mileage issue with two separate vehicles using C3 and C3X. Using free version of connect with stock release. Haven't made any attempts to troubleshoot.

EasternPA, 2024-09-08:
I bought a used Comma 3 on July 17th. It was removed from the previous owner's account before we met to make the deal. Every single drive I've ever taken with it has started in the parking lot where we met for the sale. I've run OP devel, SGOP20v2, and SP dev. Always the same.

Memberoffoxhound, 2024-09-11:
> I have the double mileage issue with two separate vehicles using C3 and C3X. Using free version of connect with stock release. Haven't made any attempts to troubleshoot. 

Please post your dongle ID at least so comma will have something to look at.

Memberoffoxhound, 2024-09-11:
> I bought a used Comma 3 on July 17th. It was removed from the previous owner's account before we met to make the deal. Every single drive I've ever taken with it has started in the parking lot where we met for the sale. I've run OP devel, SGOP20v2, and SP dev. Always the same.

Please post your dongle ID at least so comma will have something to look at.

m01001100, 2024-09-16:
> > I have the double mileage issue with two separate vehicles using C3 and C3X. Using free version of connect with stock release. Haven't made any attempts to troubleshoot.
> 
> Please post your dongle ID at least so comma will have something to look at.

Below are my dongle IDs: 

8d23badbf0a9ea4b - C3
434d76b7c51afa8a - C3X

Just noticed I'm also experiencing the arbitrary start point.

perezkris, 2024-09-30:
I am having similar issues where my Comma 3 always starts incorrectly in Galax, yet my mileage seems correct... 
My device id 9806e7635acbfc4e
Is there anything we can do?

adeebshihadeh, 2024-09-30:
@fredyshox I think this was broken by https://github.com/commaai/xx/pull/3600/files. You need to check `hasFix`. It only affects ublox since the qcom module only reports position when it has a fix.

fredyshox, 2024-09-30:
Alright, will create PR for that

jkbkster, 2024-12-03:
Same issue with 0d965213ab7ceffa

sidgestion, 2024-12-23:
same issue with 65ef8b49f9b0dd24. Pretty close to double

sshane, 2025-09-27:
This is fixed