commaai/connect #553: Not showing end of route

sshane, 2024-09-17:
https://connect.comma.ai/57048cfce01d9625/000000fc--9826dade24

RedBaron2, 2024-10-30:
I'm going to add that this is happening to me, but it isn't the last ~30 seconds. It is more along the lines of several minutes.
This [Route](https://connect.comma.ai/0d1af2138f65dbe2/0000000e--50c5327258) starts from work and is suppose to stop at a gas station, but as you will see it stops about ~0.25 miles away.
This is the [Restart](https://connect.comma.ai/0d1af2138f65dbe2/0000000f--5320d0f46e) of the previous Route that starts at the gas station.

I'm just going to mention but the GPS is way OFF for these drives according to the map shown in the window next to the video.
This shouldn't have anything to do with the missing stops?

Thanks for taking a look into this matter.

RedBaron2, 2024-11-03:
Follow-up: 
I've messaged on the Connect discord and was told that some forks had code to hide the destination / start from being shown. A person for OpenPilot said this code is NOT in OP. 
I uninstalled SP and installed OP, and can confirm that my destination is now shown in my video on Connect.
I will have to now work with the SP fork to make this a true toggle instead of the default.
Thanks for all the assistance.